thomas-v2/S7CommPlusDriver #2: Probleme beim Durchsuchen des Variablenbaums

lircy, 2023-07-05:
Hi Thomas,
       In meinen Tests habe ich festgestellt, dass, wenn PObjects VarnameList== null IOffsetInfoType_1Dim ArrayElementCount riesig ist, was zu einem Überlauffehler führt, warum, ich hoffe, Ihre Lösung zu bekommen.

thomas-v2, 2023-07-05:
Hi,
hast du noch weitere Informationen, wann das auftreten kann? Eigentlich sollte es nicht vorkommen, dass ein Bereich Variablen ohne Namen besitzt.

Bei Bugs wäre es für mich nützlich, eine Wireshark Aufzeichnung davon zu erhalten, in welche der TLS Pre-Master-Secret-Key wie in der Readme beschrieben, integriert wurde. Dabei darauf achten, dass die Aufzeichnung nicht unbeabsichtig anderweitige private Kommunikation enthält.

Das Projekt ist zur Zeit noch sehr in der Entwicklungsphase, und noch nicht wirklich praxistauglich nutzbar, Es sind auch noch nicht alle möglichen Datentypen in allen möglichen Kombinationen getestet, es kann sein, dass dadurch Folgefehler bei der Verarbeitung entstehen.

lircy, 2023-07-09:
Hi Thomas,
Ich habe mit dem Siemens SPS-Simulator (S7-PLCSIM Advanced 5.0 &S7-PLCSIM V18) getestet, das SPS-Modell ist S7-1500 und die Firmware-Version ist V2.9. Der S7CommPlusDriver-Code wird auf einem anderen Computer ausgeführt, und die Betriebssystemversion der beiden Computer ist Windows 10 Pro.

lircy, 2023-07-09:
Hi,
Ich habe den Siemens PLC-Simulator (S7-PLCSIM Advanced 5.0 und S7-PLCSIM V18) für Tests verwendet. Der PLC-Typ ist S7-1500 und die Firmware-Version ist V2.9. Der S7CommPlusDriver-Code wurde auf verschiedenen Computern ausgeführt, wobei beide Computer Windows 10 Pro als Betriebssystem verwenden.

thomas-v2, 2023-12-26:
Du könntest nochmal mit der aktuellen Version testen, ob das Problem immer noch vorhanden ist. Ich habe einiges geändert.

closed 2024-08-23 by thomas-v2