thomas-v2/S7CommPlusDriver #4: Probleme beim Browser

RitterMaximilian, 2023-10-17:
Hallo,

Meine HW ist ein Softwarecontroller S7-1507S. Ich kann mich mit der Plc verbinden. Beim ersten exploreRequest bekomme ich 281 Elemente in meiner objList zurück (FB's,FC's,UDT's,DB's,...). Soweit sieht alles richtig aus.

Beim zweiten exploreRequest bekomme ich leider nur 4 Elemente zurück, wobei beim letzen die Varnameliste leer ist. Somit gibt die Funktion Browser eine Liste mit 3 Elementen zurück. Ich habe den Code minimal verändert. (IP-Adresse angepasst und ProDiag Instanz-DBs ausgebledet, da bei diesen die Varnamliste leer ist).

Das möchte ich eigentlich erreichen:
Ich möchte den Aufbau eines UDT's von der Steuerung aus lesen.
Ist das mit der Anwendung möglich oder muss ich wissen wie Lang ein Datentyp ist um ihn zu lesen?

thomas-v2, 2023-10-17:
Grundsätzlich sollten sich alle Informationen aus der SPS online auslesen lassen. Ich habe aber bisher immer nur an einfachen Konstellationen getestet, um herauszufinden wie das mit dem Browsen und dem Zusammenführen der Informationen überhaupt funktioniert. Dabei wird auch nochmal unterschieden zwischen Structs, Benutzer-UDTs und Bibliotheks-Typen, kann sein, dass ich da noch nicht alle Daten verarbeite (Instanz DBs fallen mir gerade ein). Da gibt es auch noch einen Datenbereich von dem ich gar nicht weiß was der bedeutet (war bei meinen Programmen bisher immer leer).

Von dem was ich gehört habe, sind die Softwarecontroller auch in einigen Details anders, ich habe damit noch nicht gearbeitet. Leider lassen sich die Softwarecontroller nicht in Plcsim simulieren, sonst hättest du mir ggf. ein Demoprojekt zur Verfügung stellen können an dem ich das testen kann.

Wenn es nicht so kompliziert ist, könntest du evtl. eine SCL Quelle von dem DB/UDT zur Verfügung stellen.

Default467, 2023-10-17:
Danke erstmal für deine Antwort.
Der Aufbau der Struktur ist schon komplexer. Ich werde morgen das gleiche Projekt auf einer S7-1500 laufen lassen und schauen, ob es hier geht. Falls nicht könnte ich dir das ganze Projekt zu kommen lassen. Wie lass ich dir die Quellen bzw. Projekte am besten zu kommen?

thomas-v2, 2023-10-17:
Wenn sich recht einfach nachstellen lässt was nicht funktioniert, das Ganze am besten soweit reduzieren, dass nichts anderes mehr vorhanden ist. Also beispielsweise wenn ein Datentyp in einer UDT nicht erkannt wird, von der UDT eine SCL Quelle erstellen, und ggf. von einem DB wo diese UDT instanziert wird. Je nach Größe dann entweder direkt hier als Code einfügen, oder als Textdatei ggf. gezippt anhängen.

Default467, 2023-10-18:
Also mit einer 1500er Cpu kann ich den DB auslesen. Somit scheint es schwierigkeiten mit dem Software-Controller zu geben. Besteht von deiner Seite aus an dieser Stelle weiter zu machen?

thomas-v2, 2023-10-18:
Im Grund habe ich die Arbeit daran zur Zeit eingestellt, weil ich dafür zur Zeit keine Anwendung habe, selber zur Zeit wenig Lust und Zeit, und von anderer Seite auch kein Interesse besteht daran mitzuarbeiten. Darum auch noch mal als Hinweis, dass der jetzige Projektstand in keinerlei Weise produktiv einzusetzen ist.

Auf einen Soft-Controller habe ich keinen Zugriff um mir das selber mal anzusehen. Wenn zumindest Daten von der Steuerung zurückkommen, und diese im Wireshark Log auch sichtbar sind, dann könnte ich ggf. daran einmal vergleichen wo die Unterschiede sind. Vielleicht ist es ja nur eine Kleinigkeit.

Aber es gab schon Änderungen, da hat Siemens das gesamte Format umgestellt. Z.B. die 1200er mit Firmware 1, und Firmware 2 sind schon recht grundlegend anders, bei den ersten 1500ern auch. Da von diesen Steuerungen aber keine einzige TLS unterstützt, gibt es hier einen Vorteil, dass man diese ignorieren kann (Teile davon sind noch hier im Code, weil ich dieses Projekt einmal begonnen habe um mit den ersten 1200er zur Analyse zu kommunizieren. Da war das alles noch ohne Authentifizierung, Signierung oder Verschlüsselung möglich.

Falls es dir erlaubt ist Wireshark Logs von dem Browsen des Projekts mit der Soft-SPS und des gleichen Projekts mit der 1500er  zur Verfügung zu stellen, dann kann ich mir das zumindest einmal ansehen. Falls du diese hier anhängen möchtest, darauf achten, dass in dem Projekt was da gebrowst wird keine vertraulichen Daten vorhanden sind.

Default467, 2023-10-19:
Ich versuche so einen Log zu erstellen, aber aktuell verstehe ich folgenden Zusammenhang nicht. Wenn ich nur 1 DB im Projekt anlege und einen Durchlauf mache bekomme ich als Rückgabewert den Varnamen, DB Namen, Datentyp und Wert für jede einzelne Variable im DB. Wenn jedoch meherere DB, FB usw. in meinem Projekt sind bekomme ich als Rückgabewert ein Bytearray von dem DB. Muss ich dazu bei der Suche gewisse Parameter setzen?

thomas-v2, 2023-10-19:
Zur Zeit kommen bei z.B. bei Strings noch Bytearrays als Rückgabewert, weil dies das Format ist wie es auch auf dem Netzwerk übertragen ist. Da bin ich (unter anderem) gerade bei der Arbeit stehen geblieben. Für ein High-Level Interface muss der Benutzer dem Treiber sagen "ich will Variable ID x lesen, und diese ist vom Datentyp WString". Und der Treiber konvertiert dann das Byte-Array vom Netzwerk in den entsprechenden Zieldatentyp. Nur an der Antwort lässt sich z.B. ein Array of Byte nicht von einem String unterscheiden. Diese Umwandlungsoperationen sind noch bei etlichen mehr SPS-Typen notwendig, von denen es insgesamt theoretisch ca. 130 gibt. Für den Anwender relevant sind ich schätze mal um die 30.

Kann auch sein, dass wegen dem Bytearray noch an anderer Stelle etwas schiefläuft, aber das oben wäre eine Möglichkeit. Dann übertragen einige Steuerungen z.B. den Typ DTL nicht als Einzelwerte, sondern als (komprimierten) Byte-Blob, das wird dann bei der Ausgabe auch wie ein Byte-Array aussehen. Wovon das abhängt wann welche Steuerung welches Format schickt habe ich noch nicht herausgefunden. Meine Anzahl Test-SPSen liegt bei n=1 (Plcsim).

thomas-v2, 2023-10-29:
Hast du die Möglichkeit ein Logfile von dem Browsen mit dem Software-Controller zu erstellen? Würde mich interessieren was sich da anders verhält.

Ich habe noch ein kleines Hilfsprogramm geschrieben, was die Integration des key.log files in die Wireshark-Aufzeichnung etwas vereinfacht (zumindest wenn man das mehr als ein Mal macht): https://github.com/thomas-v2/PcapKeyInjector

thomas-v2, 2023-12-26:
Hat sich das Verhalten mit der aktuellen Version soweit erledigt?

closed 2024-08-23 by thomas-v2