thomas-v2/S7CommPlusDriver #6: Connect fehlgeschlagen S7-1518

Nick135, 2023-11-03:
Bei mir schlägt bereits der Verbindungsaufbau zu einer S7-1518 fehl.

|  |  |
| ------------- | ------------- |
| CPU | 1518TF-4 PN/DP |
| Artikel-Nr. | 6ES7 518-4UP00-0AB0 |
| Firmware | V2.9 |

Per OPC-UA komm ich auf die PLC.

```
Main - START
Main - Versuche Verbindungsaufbau zu: 192.168.214.100
Main - Connect fehlgeschlagen!
Main - ENDE. Bitte Taste drücken.
```

[image]

Per OPC-UA komm ich auf die PLC.

thomas-v2, 2023-11-03:
Hallo,

hast du OpenSSL, bzw. die notwendigen Bibliotheken installiert?

Siehe:
https://github.com/thomas-v2/S7CommPlusDriver#openssl

Nick135, 2023-11-03:
"OpenSSL in der entsprechenden Version installiert": Welche Version?

Hab jetzt die x64 Version 3.1.2.0 der beiden dll mit da zu gelegt. Will jedoch immer noch nicht.

[image]

Nick135, 2023-11-03:
Problem Gefunden:
Dll hieß "libcrypto_3_x64.dll" und nicht "libcrypto-3-x64.dll"

closed 2023-11-03 by Nick135

thomas-v2, 2023-11-03:
Kam die dll mit Unterstrich aus deiner Installation, oder selber umbenannt? Nur damit ich weiß, ob ich an der Dokumentation bzw. am Namen der dll im Code noch etwas anpassen sollte.

Nick135, 2023-11-06:
Hab die dll aus dem OpenVPN Ordner genommen. Dort liegen se mit Unterstrich.