thomas-v2/S7CommPlusDriver #10: Probleme beim Browser

Nick135, 2023-11-24:
Hallo Thomas,

ich versuche gerade die Lib zu verstehen und habe an einer SINUMERIK ONE NCU 1760 mit S7-1500 einen Datenbaustein hinzugefügt und erwartet, dass dieser durch Browsen gefunden wird. Dem ist jedoch nicht so.
Es werden jedoch Standardbausteine der SINUMERIK gefunden.

|  |  |
| - | - |
| NCU | 1760 |
| Artikel-Nr. | 6FC5 317-6AA00-0Axx |
| Firmware | V6.20 |
|  |  |
| CPU | 1500 |
| Artikel-Nr. | 6FC5 518-4FP00-6AB0 |
| Firmware | V2.9 |
| TIA | V17 |

[image]

[image]

```
List<VarInfo> vars = new List<VarInfo>();
res = conn.Browse(out vars);
// Browse res => 0
// vars.Count => 2616
// vars.Where(f=>f.Name.ToLower().Contains("plus")).Count() => 0
vars.Where(f=>f.Name.ToLower().Contains("dummy")).Count() => 0
```

thomas-v2, 2023-11-24:
Ich habe noch nie eine Aufzeichnung zu einer Sinumerik NCU im S7CommPlus Protokoll gesehen. Es könnte sein, dass hier die Datenbausteine eine andere ClassId besitzen. Bei den normalen Steuerungen setzt sich diese aux 0x8a0eXXXX zusammen, wobei XXXX dann die Datenbausteinnummer ist.

In der jetzigen Browse() Funktion wo alles aus der Steuerung in einem Rutsch abgefragt wird, werfe ich aus der Antwort alle Objekte mit einer anderen ClassId raus. Da wäre zu prüfen, ob a) die Datenbausteine in der Antwort überhaupt vorhanden sind, und b) wenn ja, welche ClassId diese besitzen.

Man könnte die Informationen auch über das Programm dumpen, aber einfacher ist das meiner Meinung nach an einer Wireshark Aufzeichnung zu analysieren. Wenn ich mir das mal ansehen soll, bitte eine Wireshark Aufzeichnung vom browsen erstellen, und wie auf der Startseite beschrieben in die pcapng die TLS Schlüssel integrieren. Wenn du die Daten hier nicht anhängen möchtest, kannst du mir diese auch per Mail zukommen lassen.

Nick135, 2023-11-24:
@thomas-v2 Hab dir gerade ne Mail geschickt.

Nick135, 2023-11-24:
DB kommt an. Dann wird es wohl am Auslesen der Variablen Namen scheitern, oder ich hab das mit "vars.Name" falsch verstanden.

[image]

thomas-v2, 2023-11-24:
Der Name kommt schon bei der ersten Anfrage zurück. Im zweiten Schritt werden dann die Typ-Informationen abgefragt, d.h. der Aufbau des DBs sowie aller Datentypen und FBs, und das ist bei dir ein ziemlich großer Datensatz. Vielleicht schlägt dort etwas fehl.

Ich habe die Funktionen zum Abfragen einzelner Bausteine schon integriert. Ich kann dir mal eine Testversion zukommen lassen, bei der das schon unter einer rudimentären GUI bedienbar ist.

Nick135, 2023-11-24:
Ich hab bei den exploreData jetzt nur den einen DB hinzugefügt.
Dennoch enthält die varInfoList nicht die Variablen des DBs. varInfoList ist jetzt null. Wireshark wieder als Mail.

[image]

thomas-v2, 2023-12-22:
Ich denke, das meiste was diesen Fall betrifft, sollte behoben sein.

closed 2023-12-22 by thomas-v2