thomas-v2/S7CommPlusDriver #13: Lesen nach Schreiben schlaegt fehl

Freax13, 2023-12-13:
Ich habe folgenden Code verwendet:
```csharp
using System;
using System.Collections.Generic;
using S7CommPlusDriver;

namespace DriverTest
{
    class Program
    {
        static void Main(string[] args)
        {
            string HostIp;
            int res;
            Console.WriteLine("Main - START");
            // Als Parameter lässt sich die IP-Adresse übergeben, sonst Default-Wert von unten
            if (args.Length == 1)
            {
                HostIp = args[0];
            }
            else
            {
                HostIp = "192.168.13.4";
            }
            Console.WriteLine("Main - Versuche Verbindungsaufbau zu: " + HostIp);

            S7CommPlusConnection conn = new S7CommPlusConnection();
            res = conn.Connect(HostIp);
            if (res == 0)
            {
                Console.WriteLine("Main - Connect fertig");

                {
                    List<PValue> writevalues = new List<PValue>();
                    PValue writeValue = new ValueUSInt(99);
                    writevalues.Add(writeValue);
                    List<ItemAddress> writelist = new List<ItemAddress>();
                    writelist.Add(new ItemAddress("8A0E0064.A"));
                    List<ulong> errors = new List<ulong>();
                    errors.Clear();
                    res = conn.WriteValues(writelist, writevalues, out errors);
                    Console.WriteLine("res=" + res);
                }

                {
                    List<ItemAddress> readlist = new List<ItemAddress>();
                    readlist.Add(new ItemAddress("8A0E0064.A"));
                    List<object> values = new List<object>();
                    List<UInt64> errors = new List<UInt64>();

                    res = conn.ReadValues(readlist, out values, out errors);
                    Console.WriteLine("res=" + res);
                }

                conn.Disconnect();
            }
            else
            {
                Console.WriteLine("Main - Connect fehlgeschlagen!");
            }
            Console.WriteLine("Main - ENDE. Bitte Taste drücken.");
            Console.ReadKey();
        }
    }
}
```

Der Code schreibt erst zu einer Variable und versucht den geschriebenen Wert zurueckzulesen.

Das ganze fuehrt zu folgendem Fehler:
```
Main - START
Main - Versuche Verbindungsaufbau zu: 192.168.13.4
S7CommPlusConnection - Connect: Verwende SessionId=0x70000C88
S7CommPlusConnection - ReadSystemLimits: m_MaxTagsPerReadRequestLimit=100
S7CommPlusConnection - ReadSystemLimits: m_MaxTagsPerWriteRequestLimit=100
S7CommPlusConnection - Connect: Benötige Zeit für gesamten Verbindungsaufbau: 672 ms.
Main - Connect fertig
res=0
S7CommPlusConnection - OnDataReceived: ProtocolVersion 0xfe SystemEvent received
S7CommPlusConnection - OnDataReceived: SystemEvent mit s7HeaderDataLen > 16, vermutlich Fehler.
S7CommPlusConnection - WaitForNewS7plusReceived: FEHLER: Timeout!
res=5
Main - ENDE. Bitte Taste drücken.
```

Die versendeten Nachrichten (siehe [capture-with-keys.pcapng.zip](https://github.com/thomas-v2/S7CommPlusDriver/files/13661471/capture-with-keys.pcapng.zip)) sehen auf den ersten Blick in Ordnung aus.

Verwirrenderweise funktioniert der Code, wenn man die Reihenfolge der Lese- und Schreib-befehle tauscht. Das wuerde ich auch eher als Indiz dafuer werten, dass die versendeten Nachrichten an sich richtig sind.

Ist diese Verhalten bekannt und kann man es umgehen?

thomas-v2, 2023-12-13:
Den Fehler kann ich bei mir auch nachstellen, mit Plcsim V18.

Ich vermute mal, das liegt entweder an der SequenceNumber oder an der IntegrityId, Bisher erhöhe ich beide Werte mit jedem Request, was ich sogar schon bis zum Überlaufwert im Dauerlauf getestet habe. Aber nach einem Schreibbefehl scheint da etwas anderes erwartet zu werden.

thomas-v2, 2023-12-13:
Könntest du noch etwas zur SPS sagen an der du getestet hast, also welcher Typ?

Der Fehlercode ist auf jeden Fall der gleiche, wie wenn ich bewusst eine falsche IntegrityId schreibe. Warum sich die Berechnung nach Wechsel einer Funktion ändern sollte ist mir noch ein Rätsel.

Freax13, 2023-12-13:
Vielen Dank fuer die schnellen Antworten!

> Könntest du noch etwas zur SPS sagen an der du getestet hast, also welcher Typ?

Die Order-Nr ist [6ES7 212-1BE40-0XB0](https://mall.industry.siemens.com/mall/en/WW/Catalog/Product/6ES7212-1BE40-0XB0) und die Firmware Version ist V4.4.
 
> Der Fehlercode ist auf jeden Fall der gleiche, wie wenn ich bewusst eine falsche IntegrityId schreibe. Warum sich die Berechnung nach Wechsel einer Funktion ändern sollte ist mir noch ein Rätsel.

Okay, ich werde am Freitag mal etwas an der integrity id und sequence number rumschrauben.

thomas-v2, 2023-12-13:
Ich habe nur immer gesehen, dass Siemens eigene Client-Anwendungen immer zwei Verbindungen aufbauen. Wobei über die eine  gelesen und über die andere geschrieben wird. In dem Fall wird die Sequenznummer und IntegrityId immer um 1 inkrementiert. So mache ich es auch, und wenn man nur Get oder nur Set macht, bzw. von Get nach Set wechselt dann funktioniert es so auch. Nur von Set nach Get zurück nicht, das ist kurios.

Freax13, 2023-12-13:
Okay, fuer unseren Anwendungsfall verwenden wir gluecklicherweise auch zwei getrennte Verbindungen zum Lesen und Schreiben. Aufgefallen ist das ganze nur in einem Beispiel Program, was ein Paar mehr Sachen demonstrieren sollte, darunter abwechselndes Lesen und Schreiben in einer Loop.

Allem Anschein nach scheint der Wechsel von Set nach Get wohl einfach nicht erlaubt zu sein.

Vielen Dank noch mal fuer die Hilfe!

thomas-v2, 2023-12-13:
Na so schnell darf man nicht aufgeben.

Es ist nämlich doch möglich, wenn man für Get und Set Befehle verschiedene Sequenznummern und IntegrityIds führt, die unabhängig voneinander inkrementiert werden. Zumindest nach erstem Test funktioniert es damit, war einfach nur ins Blaue vermutet. Eventuell lässt sich das auch über die zweite SessionId umsetzen die man bei CreateObject erhält, d.h. eine SessionId für Get und die andere für Set. Kommt aber vermutlich auf das gleiche Ergebnis hinaus.

Jetzt müsste man nur noch ausprobieren, welche Klasse von Funktionen zu einer Gruppe gehört welche die gleichen Zähler verwenden, oder ob alle unabhängig voneinander ihre Zähler führen müssen.

Zwei Sockets erhöhen mir zum jetzigen Zeitpunkt zu sehr die Komplexität.

Ich öffne den Issue mal wieder, denn so ist es auf jeden Fall ein Fehler.

closed 2023-12-14 by thomas-v2

Freax13, 2023-12-15:
> Es ist nämlich doch möglich, wenn man für Get und Set Befehle verschiedene Sequenznummern und IntegrityIds führt, die unabhängig voneinander inkrementiert werden. Zumindest nach erstem Test funktioniert es damit, war einfach nur ins Blaue vermutet.

Stimmt! Desweiteren scheint es so als muesste man nur die integrityId und nicht die sequenceId getrennt inkrementieren (zumindest funktioniert damit SetMultiVariables).

thomas-v2, 2023-12-15:
Ja das scheint ausreichend zu sein, werde ich nochmal anpassen. Die andere SetVariables Funktion scheint zumindest zu der SetMultiVariables dazu zu gehören, bei der Gruppe die eine andere IntegrityId-Zählung verwenden. Die SetVariables nutze ich z.B. zum SPS Stop/Run setzen, und wenn man die Funktion einfach mal zwischen einigen anderen Lese- und Schreibaufträgen einschiebt, scheint es so zu funktionieren. Warum das getrennt wurde will sich mir zwar nicht erschließen, aber wenn es funktioniert reicht es vorerst aus.

commit efe064e7 (2023-12-14): Fixes #13; Use different SequenceNumbers/IntegrityId for Set-Functions.

At least the Set functions like SetMultiVariables need a separate
counting of the SequenceNumbers/IntegrityId. Otherwise the request
failes. It seems it's enough if we have one separate numbering for the
set-functions, and another for all other calls.