thomas-v2/S7CommPlusDriver #14: Struct RW

aarmando73, 2023-12-19:
Hello Thomas,
I'm working with your project in order to understand how to read/write full STRUCT variables, as this feature will optimize the communication speed.
Just as a idea/tip (if you don't already discovered it by yourself) I saw that a ValueStruct can return a Not Optimized ByteArray, and for every single child item I use the NotOptimizedAccess (and eventually GetBitoffsetinfoNonoptimizedBitoffset() with bit values) as index in the above buffer to locate the value.
So I modified PValue Deserialize (add struct), PlcTag TagFactory  (add struct), built a PlcTagStruct, and add also Struct PObjects in Browse.

I'm going to test every single var Type in next weeks.

Alberto

thomas-v2, 2023-12-19:
Hi Alberto,

yes I know this, at least for DTL it's implemented in PlcTag.cs for PlcTagDTL. But without browsing the typeinformation, only with pre-knowledge of the offsets (hopefully DTL doesn't change it's type).

I don't know if there may be a disadvantage, because Siemens drivers aren't using this method. I think of the consistency of the data. If you access basic types, you can be sure, that you'll always read a consistent value.

I've made tests some time ago reading strings, which is'n possible to read consistent. My plc program was writing a string variable at one cycle 255 chars "aaa..." and in the next cycle "bbb..", then it's possible the client reads sometimes parts like "aabbb". I'd check if reading a complete struct doesn't break any other basic variable types consistency (e.g. LReal).

aarmando73, 2023-12-19:
Thank you for your explanation.
Until now I just worked with ISOTCP 1006 (S7Comm with NotOptimized DB) and every time I found consistency errors I supposed it was due to message splitting (only X bytes/call when reading/writing big data).
Now you tell me that even with a simple STRING (I suppose not so long in your test) you see consistency issues... this is weird...
Anyway I always ask my colleagues to write the PLC software (i work only with PC) so that consistency errors are not critical.
Good to take care on it.

closed 2023-12-20 by aarmando73

thomas-v2, 2023-12-29:
Just for information:
I discovered, that you can readout a complete datablock actual values, when you read the sub-id 5. I've seen that in a new datablock, the first variable always starts at id 9, so I just tried what is in the values before. Id 1 I know, this returns der RID you need for exploration.
For example, if you have a datablock 15, then ItemAddress("8A0E000F.5") will return a big blob with all data. If you know the offsets, then you can reinterpret the content to the variables.