thomas-v2/S7CommPlusDriver #24: Probleme beim Browsen

zhanyucheng0809, 2024-01-11:
Hallo Thomas, dies ist ein sehr tolles Projekt. Ich habe derzeit einen Fall, bei dem ich den Status und die Daten der Programmblöcke (FBs, FCs usw.) eines PLCs über Ihr Projekt abrufen möchte, um die Versionskontrolle der Projektprogrammierung durchzuführen. Allerdings kann ich mit der Methode „Browser()“ nicht die benötigten Daten erhalten. Welche Methode sollte ich stattdessen verwenden? Entschuldigung, meine C#-Fähigkeiten sind etwas schwach.
Danke!

thomas-v2, 2024-01-11:
Das Browsen ist für das Auslesen des Variablenhaushalts, und der zugehörigen IDs um später Zugriff auf diese Variablen zu erhalten.

Das Auslesen der Programme nutzt zumindest grundlegend die gleichen Funktionen die auch schon in der Bibliothek vorhanden sind. Letztenendes geht es darum, die passenden Zugriffs-IDs herauszufinden, und die Antwortdaten zu interpretieren. So wie ich das bisher analysiert habe, wird ein Programmcode doppelt gespeichert. Einmal der generierte Bytecode, und der Quellcode als gezippte XML-Datei mit einer sehr komplexen Struktur, aus der erst wieder aufwändig der Quellcode (z.B. in SCL) zurückübersetzt werden muss.

zhanyucheng0809, 2024-01-12:
Vielen Dank für Ihre ausführliche Antwort. Also, muss ich die Methode 'GetTypeInformation(uint exploreId, out List<PObject> objList)' verwenden, um mit 'exploreId' das von mir benötigte Programm zu finden?

thomas-v2, 2024-01-12:
Mit GetTypeInformation() lassen sich schon die kompletten Informationen abfragen.
Z.B. für einen FC mit der Bausteinnummer 100, wäre das so in der Art:
var ti = new List<PObject>();
conn.GetTypeInformation(0x8a130000 + 100, out ti);

Da bekommst du alles zurück was der Baustein so zu bieten hat, weil damit auch alle unterlagerten Elemente mit abgefragt werden.

Es könnte jedoch sein, dass ich noch nicht alle dort vorkommenden Variablen-Typen korrekt verarbeiten kann. Vor allem weil dort auch viele Elemente leer sein können.

Und vom XML wirst du auch nichts sehen, weil das wie gesagt gezippt mit einem Wörterbuch ist. Und das Wörterbuch welches bei dem jeweiligen Datensatz anzuwenden ist, muss man auch kennen.

zhanyucheng0809, 2024-01-13:
Vielen Dank für Ihre Antwort! Ich habe davon sehr profitiert

zhanyucheng0809, 2024-02-20:
Hallo Thomas, wie Sie zuvor erwähnt haben, wird der Programmcode zweimal gespeichert. Einmal als generierter Bytecode und einmal als Quellcode in einer komprimierten XML-Datei. Werden diese Daten in der ti, die von conn.GetTypeInformation(0x8a130000 + 100, out ti) zurückgegeben wird, gespeichert? Ich möchte nun ein Projekt starten, das regelmäßig automatisch Programme aus der PLC abruft und sichert, ohne die Programme zu interpretieren. Bei Bedarf soll das gesicherte PLC-Programm wieder in die PLC geschrieben werden.

thomas-v2, 2024-02-20:
Wie der Upload oder Download von Programmbausteinen im Detail abläuft, habe ich mir noch nicht angesehen. Das besteht aber aus mehr, als nur ein einziges Objekt herunterzuladen, oder hochzuladen. Es gibt auch noch einen Program-Change Counter, wobei ich nicht weiß ob der manuell angepasst werden muss, oder automatisch inkrementiert wird,
Ich würde mir das ansehen wie das aus dem TIA-Portal her ausgeführt wird, und das dann versuchen selber umzusetzen. Du kannst im TIA-Portal auch TLS für eine Sitzung abschalten, dann kannst du alles mit Wireshark wie bekannt mitschneiden und es dir ansehen.

zhanyucheng0809, 2024-02-21:
Vielen Dank für Ihre professionelle und geduldige Antwort. Ihre Hilfe ist nicht nur eine direkte Antwort auf die Frage, sondern auch eine Art der detaillierten Anleitung, die es mir ermöglicht, komplexe Konzepte zu verstehen und die Probleme, mit denen ich konfrontiert bin, zu lösen.

closed 2024-08-23 by thomas-v2