thomas-v2/S7CommPlusDriver #26: Write String and WString not working

3Philipp3, 2024-02-18:
Hello!

First of all thank you for your outstanding work, this is a really great project. I ran some tests with a CPU 1510SP-1 and I encountered some problems with writing strings. Reading is ok, for all other datatypes that I've tried writing is also ok just strings and wstrings are 
always giving me errors (S7CommPlusConnection - WriteValues: Write with errors. ReturnValue=2318368645692325907).

Thanks

thomas-v2, 2024-02-19:
Are you using the tag classes for writing, PlcTagWString / PlcTagString? The constructor has an additional parameter, where you have to pass the string max.-length. By default it's 254, this is the length if you define a WString or a String without a specified length.
This is neccessary, because on writing a string, you need to write all bytes of the string, including the header bytes.

3Philipp3, 2024-02-19:
Hello,

no, I just used it like this:

public void WriteStringVar(String SymAdress, string val)
{    
    List<PValue> writevalues = new List<PValue>();
    PValue writeValue = new ValueWString(val);
    writevalues.Add(writeValue);
    List<ItemAddress> writelist = new List<ItemAddress>();
    var v = vars.Find(x => x.Name == SymAdress);
    writelist.Add(new ItemAddress(v.AccessSequence));
    List<ulong> errors = new List<ulong>();
    errors.Clear();
    int res = Client.WriteValues(writelist, writevalues, out errors);
}

But there is just the used Stringdata in the buffer (Stringlength + chars).

I changed it now to the PlcTagString method you suggested and this is working.

Thanks a lot!

thomas-v2, 2024-02-19:
The PlcTag classes are a layer above the ReadValues / WriteValues methods, as for some Plc datatypes you need informations of how to process the read data, and how to write the data. As you have seen for example in strings, they are transmitted as an array of byte (String) or ushort (WString).

3Philipp3, 2024-02-24:
Thank you for that information.
Really appreciate your work on this.

closed 2024-08-23 by thomas-v2