thomas-v2/S7CommPlusDriver #29: Unable to decrypt packets

buffyslays, 2024-06-04:
I am attempting to decrypt Wireshark packets between the S7CommPlusDriver and a S71200. 

I have tried both ways that you have including in your README. 

**1. Place the log file in a directory and make it known to Wireshark. To do this, go to the Wireshark menu → Settings. Under Protocols, select TLS, and select the appropriate file in the (Pre)-Master-Secret log filename field**

I point WireShark to the key log file that was created in the bin file when I captured the data from the program (edit -> preferences -> TLS -> Pre-Master-Secret log filename). I save the file as S7CommPlusTest.pcapng and close WireShark. I re-open the file and nothing has changed.

**Integrate the secrets directly into the Wireshark recording**

I have tried this manually using the command prompt as well as using the Pcap Key Injector utility tool included in your project. When I open the S7CommPlusTest_withKey.pcapng file the packets are still encrypted. I have verified that I am using the correct key log file with each capture. I have verified that the version of WireShark I am using is the latest and includes the Siemens dissector. 

I have tried this on two different PC's and am unable to view decrypted packets. Is there something that I am missing?

thomas-v2, 2024-06-05:
You should see als in the Wireshark capture without keys, at least the S7COMM-PLUS "Req InitSSL" and "Res InitSSL". After this the TLS enrcryption is handshaked and then active. Do you see these packets in your captures? You need to start the capture before starting the comminication, so Wireshark can see the TLS handshake packets.

buffyslays, 2024-06-05:
Hi Thomas, thanks for the quick response. 
We were confused thinking the S7COMM-PLUS was bundled into Wireshark along with S7COMM! We added the plugin.dll and we are in business. Wishing I hadn't made that assumption. Thanks for pointing out which packets to look for!

thomas-v2, 2024-06-05:
The hint that you need the plugin dll for S7comm-Plus is in the readme. Yes it's a bit confusing that S7comm is integrated, and S7comm-plus not (not now).

closed 2024-08-23 by thomas-v2