thomas-v2/S7CommPlusDriver #38: PLC：System initiated session termination

Dstary, 2024-08-21:
Hello thomas，When I used the project to connect the PLC and read the data once per second, an error occurred after about 10 minutes。A previously established system session was terminated due to CPU operating system security efforts.
[image]

thomas-v2, 2024-08-21:
I'm mostly testing with Plcsim, maybe the behaviour is a little bit different there. I guess it's related to TLS re-keying, if I remember correctly, I've had some problems with this first but hoped it was fixed.
I try to reproduce it with Plcsim, as I have no real Plc available for now.

thomas-v2, 2024-08-23:
I've run a test with Plcsim over 1 hour, reading 2000 times 2500 variables, without problems or entries in the diagnostic buffer.

Do you lose the connection every 10 minutes, or do you only get the entries in the diagnostic buffer?

Is it possible for you to provide a wireshark capture of the communication, including the keys?

thomas-v2, 2024-08-25:
Regarding this Siemens FAQ article:
https://support.industry.siemens.com/cs/gb/en/view/109826658

the entry occurs when the client terminates the connection. Do you have an instable WLAN or other type of connection to the plc?

I read in other posts, that Siemens has "improved" the entries in the diagnostic buffer with newer firmware version. If you can give the model of your plc and the current firmware version, we can check if there's an improvement in newer versions.