thomas-v2/S7CommPlusDriver #41: OpenSSL connection error

stinger0698, 2024-09-16:
For some reason, when i run communication with PLC as windows service, after couple minutes it throw following error. 
[image]
For some reason it cant find function in libcrypto-3-64x.dll. Perhaps someone has faced such a problem. Although, the console version works without that error.

Wireshark log:
[s7CommPlus.zip](https://github.com/user-attachments/files/17011013/s7CommPlus.zip)

thomas-v2, 2024-09-17:
Maybe when running as a service, your application doesn't search in the application directory, but in the standard locations. For a test, you can try to add the path where the dll is located to the system variable "PATH". Copying in to windows/system32 path would also be an option, but to dump dll's there without a install/deinstall routine is not a good way.

stinger0698, 2024-09-18:
Well i tried to check libcrypto-3-64x.dll export table with dumpbin and it seems 'BIO_should_retry' is not in it.
[image]
Perhaps it is possible to handle error in some other way at SendPendingData() at OpenSSLConnector.cs?

thomas-v2, 2024-09-18:
You're correct, BIO_should_retry is in the bio.h only a define for another method with a defined parameter. I don't know why this never occured as error. I'll check for other defines which I haven't seen, and make a change.

thomas-v2, 2024-09-19:
You can try if the latest change fixes also your problem.

stinger0698, 2024-09-24:
Thank you, worked for couple days without any problem.

closed 2024-09-24 by stinger0698