thomas-v2/S7CommPlusDriver #43: Errors during continuous tag reading

stinger0698, 2024-10-24:
I am using "ReadValues(List<ItemAddress> addresslist, out List<object> values, out List<UInt64> errors)" method in infinite loop to read tags and sometimes i get following errors:

1. Unhandled Exception: System.NullReferenceException: Object reference not set to an instance of an object.
     at OpenSsl.OpenSSLConnector.GetNextBuffer(LinkedList`1) + 0x11
     at OpenSsl.OpenSSLConnector.PerformWrite() + 0x15
     at OpenSsl.OpenSSLConnector.RunSSL() + 0x73
     at S7CommPlusDriver.S7Client.RunThread() + 0xd7
     at System.Threading.Thread.StartThread(IntPtr) + 0xee
     at System.Threading.Thread.ThreadEntryPoint(IntPtr) + 0x1a
2. "WaitForNewS7plusReceived(int Timeout)" in ReadValues method return res=5 (Got this error when tried to connect to plc from two processes simultaneously too). 

Is ReadValues method is not fit for fast data acquisition? And do you have any information about ather res error codes, to improve errors handling?

thomas-v2, 2024-10-24:
For 1:
I've run a test for one hour, reading 2000 times 2500 variables as fast as possible. Means doing ReadValues immediately after the last response. The SSLConnector memory handling is a bit "stinky", I was happy when I got this part running, but It needs for sure some work. Do you use something special, like multiple threads or something, which I can try to reproduce?

For 2: 
Return value 5 (S7Consts.errTCPDataReceive) is returned when the receive timeout is elapsed, The default value is 5000ms.

stinger0698, 2024-10-25:
Thank you for reply. It seems both errors are resolved, i just added Thread.Sleep() after read values. Both errors occured in first couple of minutes, after program start, perhaps it can be some network problems.

1. Single thread and only 3 variables, using same method as you described, first getting ItemAddress then ReadValues in loop as fast as possible. And i converted your library to .net8, but i don't think it cause problems with OpenSSL.
2. After res = 5 error it sometimes just stucks somwhere in code, and do nothing. Didn't have possibility to debug yet.

I will try to reproduce and debug errors, when I will have access to a PLC.