thomas-v2/S7CommPlusDriver #45: Error handling

buffyslays, 2024-11-07:
Hi Thomas,
Where are the error codes from the S7Consts class from? Are they S7 specific error codes?

What about other error codes such as:
 ServiceNotAllowed 0xe20194000c46ff95
InvalidAID 0xe20188000f20ffe0

It does not look like you are handling these?
I need to be able to handle these as well, do you know where I can find a list of all possible  S7CommPlus error codes? 

I'm also wondering why you mention that the errors need to be handled at a higher level:

(In the deserialize methods)
 if ((ReturnValue & 0x4000000000000000) > 0) // Error Extension
            {
                // Decode the error object, but don't use any informations from it. Must be processed on a higher level.
                S7PlusObject errorObject = new S7PlusObject();
                ret += S7Plus.DecodeObject(buffer, ref errorObject);
            }

Why not just throw an exception here when the returnValue is equal to an error code?

thomas-v2, 2024-11-08:
The errorcodes in S7Consts.cs are from SharpS7, where I use basic TCP/ISO socket class from. The error codes only from the TCP and ISO parts are valid here.

The errorcodes from the S7commPlus protocol, are when my interpretation is correct, diveded into 8 parts from the 64 bit value using bitmasks. You can find some values in the sourcecode of my Wireshark dissector plugin:
https://sourceforge.net/p/s7commwireshark/code/HEAD/tree/trunk/src/s7comm_plus/packet-s7comm_plus.c

I haven't decided whether to use exceptions or return values. Also I haven't completely figured out, at exactly what errors you need to do a complete reconnect, and at what errors you can proceed with whatever you want to do.

For example, at the first CreateObject to start the session, you get always a return value > 0, it's not an error at this point.

buffyslays, 2024-11-08:
ok, thanks for the info! Looking through the dissector code I can see two different error code arrays:
1. static const val64_string errorcode_names[]
2. static const val64_string genericerrorcode_names[]

As far as I can tell, the genericerrorcode_names are not being used, is this correct? If they are being used, how do we know when to use errorcode or genericerrorcode?

Thanks again for all your hard work and quick responses, very helpful.

buffyslays, 2024-11-08:
OK, good to know! Thank you for all your hard work and quick responses. Very helpful.

closed 2024-11-08 by buffyslays

thomas-v2, 2024-11-08:
Just use Wireshark with the plugin and analyze some communication. In the Wireshark UI you can easy see how the returnvalue part is divided. I've written the Wireshark dissector first, and then / or parallel the driver, so you can find many parts in the Wireshark dissector in this code.