thomas-v2/S7CommPlusDriver #48: Inconsistent data type with PLC

Siriling, 2024-12-20:
Hello, I've discovered an issue while using it. Specifically, there's a discrepancy between the data types in the PLC and those obtained through this driver.

I can check the data type serial number through the following code:
```c++
private void treeView1_AfterExpand(object sender, TreeViewEventArgs e)
{
......
    for (int i = 0; i < pObj.VarnameList.Names.Count; ++i)
    {
        Console.WriteLine("0:{0}", pObj.VarnameList.Names[i]);
        Console.WriteLine("1:{0}", pObj.VartypeList.Elements[i].Softdatatype);
......
}
```

[image]

The correct data type number should be 131

```c++
public const uint S7COMMP_SOFTDATATYPE_EVENTHWINT = 131;
```

thomas-v2, 2024-12-20:
Confirmed, there's something wrong with the type id for at least this type.

Siriling, 2024-12-23:
The issue has been resolved after updating the code. Thank you very much!

closed 2024-12-23 by Siriling