thomas-v2/S7CommPlusDriver #49: Support lower versions of firmware and variable annotations

cpf083, 2024-12-23:
hello Thomas，S7 COMM Plus is excellent, but I personally feel it has too many restrictions regarding the firmware requirements for PLCs. Both ibaPDA and Kepware software support lower versions of firmware and can also retrieve variable comments. These two software solutions are currently available for free trials. Is it possible to use Wireshark to capture the communication packets of these two software programs to address these two issues?

thomas-v2, 2024-12-24:
I know how to read the variable comments, the basic function I have added some weeks ago, To get the comments, you must read the information by an XML dataset, which also includes all the informations like the IDs we browse with the current method. But to combine both methods is not that easy, it would be easier to skip the current method and retrieve all informations from the XML dataset. I haven't compared both methods, but I guess the current method would be faster and creates less traffic.

Supporting non-TLS firmware version: It's in my opinion problematic in Open Source software. Closed software can hide where they got their informations from.

jogibear9988, 2025-01-23:
@cpf083 you could try to combine https://github.com/bonk-dev/HarpoS7 with this library

Deepak14041507, 2025-10-15:
> Wich firmware has your PLC?

Firmware version is V4.5