thomas-v2/S7CommPlusDriver #52: Meldungen in mehreren Sprachen empfangen

kkranich97, 2025-01-23:
Ich versuche die Meldungen einer S7 1500 Steuerung sowohl in Deutsch als auch in Englisch zu empfangen, leider funktioniert es nur in Deutsch. Ich habe den Parameter in der AmarmsHander.cs wie Folgt angepasst:
`asrefsobj.AddAttribute(Ids.AlarmSubscriptionRef_AlarmTextLanguages_Rid, new ValueUDIntArray(new uint[2] { 1031, 1033 }, 0x20)); // Empty for all languanges? Otherwise e.g. 1031 for de-de or what you need.`

Wenn das Array leer ist, hat es leider auch nicht funktioniert. Wie sieht es aus, wenn zwei Sprachen übermittelt werden, sollten sich in dem Notification Object dann zwei P2Objects befinden? 

Weiterhin habe ich versucht, mit dem Beispielcode die Alarmliste auf Englisch auszulesen, dabei waren dann alle Texte leer:
`var alarm = _connection.AlarmSubscriptionCreate();`
`var alarms = new Dictionary<ulong, AlarmData>();`
`_connection.ExploreASAlarms(ref alarms, 1033);`

Hat jemand eine Idee, was ich falsch gemacht habe, bzw. hat es schonmal jemand geschafft, die Meldungen in mehreren Sprachen zu empfangen?

thomas-v2, 2025-01-23:
Ich finde gerade kein Logfile wieder, ob und wie ich das mit der Mehrsprachigkeit damals getestet habe. Ich habe das jetzt nochmal erneut getestet, und das scheint etwas anders zu funktionieren.
Es kommen so wie es aussieht bei einer Alarm-Notification wenn man die Texte mit gesendet haben möchte, auch immer alle Texte in allen konfigurierten Sprachen mit. Unabhängig davon, was beim Anmelden der Subscription eingetragen wurde (oder das ist bei mir aktuell ein Bug beim Test mit Plcsim).

Auf jeden Fall geht mir gerade ein Licht auf, was die Zahlenwerte in der switch/case in der Methode FromNotificationBlob() bedeuten.  Die 67567617 ist in hex 0x04070001. 0x0407 ist 1031 dez, und damit die LCID für de-DE. Das low Word ist die Durchnummerierung der Textids (1=Infotext, 2=Alarmtext, 3=Zusatztext1 usw.). Wenn man als Sprache zusätzlich englisch en-US projektiert, was eine LCID von 1033 aufweist, dann beginnen die Keys entsprechend bei 0x04090001 bzw. der Meldetext ist dann 0x04090002.

Der Key ist also 0xAAAABBBB, mit AAAA=LCID und BBBB=TextId. Benutzer die in TIA nicht in de-DE projektieren, dürften also aktuell überhaupt keine Meldetexte mit dieser Methode angezeigt bekommen.

Ich versuche am Wochenende da mal etwas anzupassen. Ich würde der Funktion FromNotificationBlob() dann einen Parameter für die LCID ergänzen. Wenn du mehrere Sprachen extrahieren möchtest, müsstest du diese dann mehrfach aufrufen.

thomas-v2, 2025-01-24:
Ich hab zumindest schon mal ergänzt, dass die angegebene Sprache korrekt ausgewertet wird. Kommt das vor, dass gleichzeitig mehrere Sprachen auszuswerten sind? Bezüglich Netzwerklast würde ich dann auch überlegen, vorher die Texte auszulesen, und dann in den Alarmereignissen nur noch die IDs übertragen zu lassen. Sich die Texte gleich fertig mitschicken zu lassen, besitzt aber den Charme, dass Textlisten oder Begleitwerte gleich fertig in den Meldetext eingebaut werden, und man nicht anschließend wieder alles aus Einzelteilen zusammensetzen muss.

jogibear9988, 2025-01-25:
> kommt das vor, dass gleichzeitig mehrere Sprachen auszuwerten sind?

Bein der Visualisierung in unserer Firma (nutzen dort AGLink), kommt es schon vor das man Meldungen in mehreren Sprachen aufzeichnet. Wir haben diese Bspw. mehrsprachig, und wollen natürlich auch im Archiv später zur Laufzeit die Sprache umschalten können.
D.h. wir müssen alles archivieren.

kkranich97, 2025-01-29:
Hallo Thomas, habe es gerade erfolgreich getestet. Danke

closed 2025-01-29 by kkranich97