thomas-v2/S7CommPlusDriver #61: Need help generating full 56-byte Encrypted Challenge in S7CommPlus session setup

lircy, 2025-06-10:
Below is a concise GitHub issue drafted in English to seek help with generating the encrypted challenge:

---

**Title:** Need help generating full 56-byte Encrypted Challenge in S7CommPlus session setup  

### Context  
I'm implementing the S7CommPlus protocol (based on your German project) to connect to Siemens S7-1200/1500 PLCs without TLS. The session setup requires generating a **56-byte encrypted challenge** in `SetSessionSetupData`.  

### Current Progress  
- First 16 bytes of `encryptedChallenge` are correctly computed by XORing the PLC's `challenge` (bytes 2-17) with a fixed parameter `5f cb e9 73 01 70 f3 37 c9 fb 55 1a 34 32 14 ea`.  
- **Problem:** The remaining 40 bytes (positions 16-55) remain unresolved. Currently, 24 bytes are hardcoded (`3e c8 d4 a9 ... cb 57`) and the final 16 bytes (`82 6a ... 28 87`) appear to be a checksum.  

### Code Snippet  
```csharp  
// Relevant portion of SetSessionSetupData  
byte[] encryptedChallenge = new byte[56];  
byte[] para = new byte[] { 0x5f, 0xcb, 0xe9, 0x73, ... , 0xea };  
byte[] plainText = new List<byte>(challenge).GetRange(2, 16).ToArray();  

// Correctly computes first 16 bytes:  
for (int i = 0; i < plainText.Length; i++)  
{  
    bytes[i] = (byte)(plainText[i] ^ para[i]);  
}  
Buffer.BlockCopy(bytes, 0, encryptedChallenge, 0, 16);  

// Remaining 40 bytes UNKNOWN (currently hardcoded):  
Buffer.BlockCopy(new byte[] { 0x3e, 0xc8, ... , 0x57 }, 0, encryptedChallenge, 16, 24);  
Buffer.BlockCopy(new byte[] { 0x82, 0x6a, ... , 0x87 }, 0, encryptedChallenge, 40, 16);  
```  

### Request  
Could you clarify:  
1. What algorithm or keys derive the **24 middle bytes** (positions 16-39)?  
2. How is the **final 16-byte checksum** (positions 40-55) computed?  
3. Is there a cryptographic primitive (e.g., AES-CBC, custom cipher) involved?  

Any insights, pseudocode, or references would be invaluable. Thank you!  

---  

### Key Points  
- Clearly states what works (first 16 bytes).  
- Explicitly identifies the unknown parts (40 bytes).  
- Asks specific technical questions about the algorithm.  
- Maintains a respectful tone while acknowledging the original work ("based on your German project").

jogibear9988, 2025-06-10:
@lircy 
I could not help you with your question (didn't dig into the protocol yet), but maybe the project harpo (https://github.com/bonk-dev/HarpoS7) could help you. As they already support connection to older CPU's.

jogibear9988, 2025-06-16:
@lircy did you get any further ?

lircy, 2025-07-01:
There has been no progress so far.

lircy, 2025-07-04:
Here's the precise technical translation:
However, one issue remains unclear: How does the open-source project you recommended obtain the public keys for various Siemens PLC models, and is the acquisition method legal? This directly impacts future feasibility of adding support for new PLC models.

jogibear9988, 2025-07-04:
The question is, is it illegal to obtain it via reverse engineering. I think in germany it is allowed to get them for achieving interoperability.

lircy, 2025-08-06:
lircy/S7CommPlusV3Driver: This communication driver is developed on the .NET Framework 4.7.2. It enables communication with Siemens S7-1200 and S7-1500 PLCs without enabling TLS scenarios. The driver supports: Online browsing of the global symbolic variable tree. Batch reading of variable values via symbolic variable names. Writing values via symbolic variables.<https://github.com/lircy/S7CommPlusV3Driver>
[https://opengraph.githubassets.com/1fd3281787c2581ec6d3966e21c7d5951d9afdecdf606c0b166a443ab988cbbc/lircy/S7CommPlusV3Driver]<https://github.com/lircy/S7CommPlusV3Driver>
GitHub - lircy/S7CommPlusV3Driver: This communication driver is developed on the .NET Framework 4.7.2. It enables communication with Siemens S7-1200 and S7-1500 PLCs without enabling TLS scenarios. The driver supports: Online browsing of the global symboli<https://github.com/lircy/S7CommPlusV3Driver>
This communication driver is developed on the .NET Framework 4.7.2. It enables communication with Siemens S7-1200 and S7-1500 PLCs without enabling TLS scenarios. The driver supports: Online brows...
github.com