thomas-v2/S7CommPlusDriver #62: Performance issue with ReadTags (Slow Reading)

SimoneMaffei98, 2025-06-11:
Good morning,
I’m using this library to read variables from a Siemens S7-1513 PLC with firmware version 2.9.
There are about 11,000 variables to read. It’s not an excessive amount, but when I try to read all of them at once using the ReadTags function, the execution takes more than 3 seconds, which is unacceptable considering the expected time should be around 10 ms.

After further investigation, I found that the issue seems to be within the SendS7plusFunctionObject and WaitForNewS7plusReceived functions, which are internally called by ReadValues. Each call takes around 20–30 ms, and since the function splits the read requests into chunks of 100 variables, it's invoked approximately 110 times.

Is it possible to send a single request to retrieve all values at once?

Using Wireshark, I compared the network traffic with another similar application that seems to use the same functions. While we are sending multiple requests to read the 11,000 variables, the other application appears to send a single request and then receives the data in around fifty consecutive packets in just around 15ms. Unfortunately, I couldn’t analyze the content further since the communication was encrypted.

Is it possible to achieve a similar behavior using this library?

Thank you very much.

thomas-v2, 2025-06-11:
The GetMultiVariables method allows only the maximum number with one request, which is a system limit of the CPU and is read after connection setup and used later. If this is 50 and you'll add 51 to the request, then the complete request will fail.

The only option to make this faster I know, is to use Subscriptions up to the limits of the CPU, and if the limits are exceeded then switch for the rest to poll mode and read them via GetMultiVariables. This is the way for example WinCC is doing this. For Subscriptions I've done only basic tests on this type of communication, so it's not ready to use yet.

But the request either for reading or creating subscriptions for 11.000 variables will take in the best way about 121.000 bytes. With a TPDU size of 1024 which is always relevant, this will take a minimum of 120 TDPUs. So this can't be a single request PDU to access this amout of data.

There's also another read method which I've seen, but don't know exactly which advantages it has.

To see how they are doing this, I'd would be nice if you have the option to create an unencrypted Wireshark capture of the communication. This can be done with a TLS man in the middle, or the easier way would be to use an older CPU, or just projecting an older firmware inside the TIA project which doesn't support TLS would be enough.

SimoneMaffei98, 2025-06-12:
Hi Thomas,
the subscriptions seem to be working quite well. However, I have a problem: my 11k variables might not be enough, so I'm trying with 100k. The issue is that a single subscription can handle a maximum of 40k variables, so I need to create more than one.
It doesn't seem possible to do this through the same connection — do you know if there's a way to achieve this without having to create multiple connections?

jogibear9988, 2025-06-12:
One subscription can handle 40k variables? If you register them, do you get all the values after the subscription and after that only the changes?

SimoneMaffei98, 2025-06-12:
Basically yes, this is what I do:

[image]

I think it is something in test mode so you will need to change something in TestWaitForVariableChangeNotifications.

SimoneMaffei98, 2025-06-13:
Is there a way to make a bulk request to retrieve all the values at once and then convert them in the code?
100 ms still seems too much to me, and I’d like to achieve something around 10 ms, as I already do with Sharp7.

jogibear9988, 2025-06-13:
You could read 40k from a S7 with the old protocol in 10ms? That seems nearly unbelivable

thomas-v2, 2025-06-13:
If you enable Put/Get communication in the HW config of the Plc, you can still use Sharp7 to access the data. The datablock must be "not optimized". You can access this areas using absolute addressing also with the new protocol, but I guess it's always slower as the old protocol (due to more protocol overhead and TLS encryption).

thomas-v2, 2025-06-13:
But there seems to be an option, to read out a complete datablock data area in a whole dataset. I've just checked what TIA is doing when you go online in a datablock view. There is at least a subscription created, where a blob is transmitted with the raw data, without any overhead. But the subscription is created different from the standard variable subscriptions. There is first a Tis (Test and Installation) subscription created, and then another subscription where the first is referenced to. There are many unknown values in this, so I don't think this would be possible to do in the same way.

SimoneMaffei98, 2025-06-17:
This is exactly the solution I was looking for. It's important to bind the value correctly to the variable, but this approach seems to be very fast. Thank you.

thomas-v2, 2025-06-17:
Calculation the offsets for simple tags or tags in structs is relatively easy. But if there are arrays, this needs knowledge of the size of the variable. For array of structs you must known the size of one element, to know at which offset the next element starts. Maybe there are some padding bytes added at some point. This needs to be tested, and hopefully it's always the same padding style for all controllers and all TIA versions.

jogibear9988, 2025-06-17:
Was it not so that TIA also keeps the order of objects even if they are moved later? So if you insert an element in a struct you don't loose data?
So you can not relly calculate it you need to have the project.
But I could be wrong here

thomas-v2, 2025-06-19:
If you read the complete DB content, you also get a modification timestamp with the data. If I would do this in this way, I'd read the complete DB first and save the timestamp, then browse the DB to get the offsets, and then start cyclinc reading the complete DB. If the timestamp has change since the first read, then the offsets maybe wrong and must be read from the plc again.

Reading this way is not as safe as reading single variables, but if someone needs this for high speed data aquisition of many variables, this can be the only way do this.

I'm interested in this, because you can create a memory map to so how the variables are organized in the Plc, and especially what and when is something changing when doing modification in the DB (name changing, adding something, etc.)

SimoneMaffei98, 2025-06-20:
Hi Thomas,
I took some time to better analyze the situation, and you’re actually right: if variables are moved within the same DB, issues arise with the relationship between what the Browse returns and the byte array of the values.

However, it seems possible to find the variable-value mapping using @jogibear9988's function, which returns the DB’s XML structure. In that format, the variables are listed in the order they are declared, making it possible to establish the correct association.

The DB timestamp is also very useful to determine whether the DB has been modified. This way, we can trigger an automatic re-browse only if the DB has actually changed.

[image]

[image]

thomas-v2, 2025-06-20:
All informations to get the offsets are already available with the current Browse method. I only need to workout to calculate the array offsets, but the size of an element is also transmitted.

thomas-v2, 2025-06-20:
I don't know what you want to show with the pictures. There is no need to rely on any order, as the offsets (optimized and not-optimized, and also the bitoffsets) are directly transmitted, relative to the parent offset.

thomas-v2, 2025-06-21:
I have the calculation ready, but there are some strange things going on, where I don't know if this is a bug in Plcsim (V18) where I'm testing with:
- The format of the modification timestamp which is returned, is in a not known format and completely different from other uses. I've made some tests changing the DB, the value changes, but I can't reproduce what the values are. I think it's a bug in Plcsim.
- A Bool in an "optimized" DB takes one byte in memory. But multiple Bool in a system datatype like IEC_COUNTER, IEC_TIMER etc. are in a single byte with bitoffsets
- If you create an "optimized" DB, you can get also the "not optimized" offsets. They are there, even when TIA needs a compiletion to show the offsets.
- Bitoffsets in Bool-Arrays in "not optimized" cannot be browsed but must be calculated (haven't sone this yet)
- If you read the complete DB, you also read the bytes entered at "Memory reserve" in "Download without reinitialization"
- The bitoffsets in a "not optimized" DB are transmitted in a different location of the network protocol, dependant of if they are located in a struct or not

I don't wonder about the bugs in TIA portal, when I see the mess about the special cases with the "optimized" and "not optimized" access.

I think this would need a little testsuite, to compare the values read by tags and read from the offsets.

jogibear9988, 2025-11-21:
Means in your version it is faster? What have you changed?

lircy, 2025-11-22:
Performance optimizations for packet throughput were achieved through targeted code modifications in the MsgSocket.cs and S7Client.cs source files. The engineered versions of these components are provided in the accompanying attachment.

[Modify.zip](https://github.com/user-attachments/files/23686327/Modify.zip)

lircy, 2025-11-22:
The hybrid approach of combining subscription-based monitoring with active polling for reading PLC values is indeed an effective solution. However, the minimum configurable scan cycle for subscriptions appears to be limited to 100ms. Additionally, the maximum number of PLC variables that can be subscribed in a single operation varies by PLC model. For instance, in my testing, the CPU 1517 can handle up to 40,000 variables, while the CPU 1511 falls short of even 10,000 variables.

lcoeli, 2026-04-10:
> Thank you! After using this modification file, the efficiency of reading the PLC tags has been significantly improved.

You replace those 2 files? I did not see any difference here reading 5k tags.