thomas-v2/S7CommPlusDriver #63: System, Diagnostic Data and Function Call

SimoneMaffei98, 2025-06-12:
Hi Thomas,
I'd like to be able to read the system variables for diagnostic purposes. These variables seem to be defined in the library as 'Ids' or 'LID'.
The difference between them and how they can be used/read is not very clear.

This is an example used in the code:
[image]

But for example, in my case, I want to read these variables (or any other system variables):
[image]

It’s not clear how I should do this and what values I need to assign to AccessArea, AccessSubArea, and LID.Add(2);.

Is this written somewhere or am I missing something?
Is it also possible to view the value of these variables and any related addresses directly from the PLC?
How did you find the correct values associated with each variable? Is that something we can also do?

</br> </br> </br>
A similar issue applies to PLC function calls, like symbolic Browse:
[image]

Is there a manual, a list, or a way to know all the available function calls and what they do?

Thanks a lot for your help.

SimoneMaffei98, 2025-06-13:
Hi Thomas,
yes, I tried using that function, but I’d like to understand how it works completely so I can be independent in the communication and try out all the different features.

thomas-v2, 2025-06-13:
Just analyze existing communication with Wireshark, and rebuild this with the library. Or just try out, there's no documentation.

SimoneMaffei98, 2025-06-16:
Ok, thank you for all the informations

closed 2025-06-19 by thomas-v2