thomas-v2/S7CommPlusDriver #64: GetCommentsXml Language and Association Problem

SimoneMaffei98, 2025-06-12:
Hi Thomas,
I'm facing two issues when reading comments from the PLC using the GetCommentsXml function:

1. It seems that I can only read comments in English. I tried adding another language and modifying it via Tools > Project Languages, but I can only retrieve the English comments.

2. The XML returns comments with a Path (which always seems to start from 136). How can I associate a comment with its corresponding variable? The browse methods don't provide the variable's path.

[image]

Thanks.

SimoneMaffei98, 2025-06-13:
I saw that you managed to retrieve the variable ID so that the comment is linked to the correct variable. However, I still can’t see the different languages — only the English one appears.

jogibear9988, 2025-06-13:
I‘ll look… we only have one lang at the plc, i‘ll check if I could receive more

closed 2025-06-21 by thomas-v2

SimoneMaffei98, 2025-06-30:
Hi @thomas-v2 and @jogibear9988 ,
I'm trying to create a Browser using XML files. I have no issues with structures or user-defined UDTs, as I get the full hierarchy with all nested variables. However, for system-defined UDTs (such as IEC_TIMER, IEC_COUNTER, etc.), I only receive an external reference. How can I use that reference to access the structure or XML file of the corresponding system UDT?

[image]

thomas-v2, 2025-06-30:
I don't know which code you are using, as in the library there isn't a method for reading the interface description.

Maybe there's like your are reading (explore) the type information without xml for DB1 for example, using 0x92000000 + 1. And for reading the information as xml you need 0x8a0e0000 + 1. For datablocks you can get the RID from 0x8a0e0001 reading ID 1. But I don't know any method to get this backwards.

What you can try is using explore and to loop at least the first byte 0x##00001f with ## from 0x00 to 0xff and see if you get data back with any information.

And I'd try first to read only one UDT xml desciption and not a system type. From UDT  it's at least proven that there is any xml information available from your screenshot.

jogibear9988, 2025-07-02:
Or your software keeps a list of system data types, cause there interface will not change?
In the Project files you could read them, but I would think this is not necessary to keep these infos in the CPU

SimoneMaffei98, 2025-07-02:
@thomas-v2  Unfortunately, I also tried and didn’t succeed. The problem with keeping the structure in the software is that in the XML information there is also a version code. If the UDTs change between firmware versions or for other reasons, you would always need to keep a history associating structure and version, which would be complex and not optimal.

@jogibear9988  Regarding reading the project files, unfortunately I don’t know how to read them or understand how the information inside is structured.

thomas-v2, 2025-07-04:
What you can try, is to make a small project which contains only one DB with an IEC_TIMER instance. Then set the session to unencrypted connection, start wireshark, and make a capture uploading the project from the CPU into a blank TIA project. Then check in Wireshark if you see any XML contents with the structure of the Timer.

If there's no solution, I'd create XML for the system datatype for myself, from browsing the contents only for this types with the non-xml way. I don't believe the comments for these datatype will change, so you your keep then fixed. But I'd everytime browse the offsets if you need them, because there may be differences between the Bool types used, or the packing between CPU types, escpecially if you'd also like to support 1200 controllers.