thomas-v2/S7CommPlusDriver #65: Reading tags without browsing

csharpsithlord, 2025-06-19:
Is there a way to read the tags from the tag name rather than browsing everything first?

csharpsithlord, 2025-06-19:
That's what I thought but wanted to ask to be sure. This is a really good library and thanks for the quick response.

thomas-v2, 2025-06-19:
But it's possible that after many changes of the Plc program / TIA versions, the ID you once browsed is now a different variable. The protocol has the option to use a symbol CRC which is created from the symbol name and the datatype, but I'm not using this option for now. But the case that the ID is completely wrong, is a rare case, as if changing the variable name it gets a new ID and this ID isn't used again until any major change like updating to a new TIA version is done (from my experience).

csharpsithlord, 2025-06-20:
I was curious about this. Does that read all DBs / structs and ignore Merker, Input, Output, etc? Or will that read just in that DB / struct and ignore the others? I intend to populate a list / collection then update tags from that and periodically run a task to update this data for changes. The less to parse the better for speed though.

Ich-Eben, 2025-06-20:
It will just read the DB / structs that are necessary to calculate the full address of the given symbol and ignores everything else. But it also buffers the received type information in the background. So if you fetch multiple tags that are located in the same DB and or structs it calculates the address for those from the buffered type info instead of sending another browse request to the PLC.

You can call the `getPlcTagBySymbol` multiple times to populate a list. It will automatically fetch only the necessary type info for you in the background. After that simply pass the list as an array to the `ReadTags` function every time you want to update the tags values.

But keep in mind that, as Thomas said, if someone does a major change to the DB on the PLC the calculated addresses may become invalid.