thomas-v2/S7CommPlusDriver #68: Reading a large String array leads to a timeout

Stiwa-PohG, 2025-07-29:
Hello,

during my tests when reading strings, I came to the following conclusion. If the total size of all strings to be read exceeds approximately 1000 bytes, a timeout occurs during the read operation. This attempt was performed in the S7CommPlusGUIBrowser.

I tried the following READ tests:
* Successful
  * 3x String with a length of 254 characters
  * 9x String with a length of 100 characters
  * 10000 INT values
* Failed (Timeout)
  * 4x String with a length of 254 characters
  * 10x String with a length of 100 characters
  
What could be the reason for the timeout when reading multiple strings?
Thank you.

PLC Tag String Structure
[image]

Hardware Used
[image]

bgattinger, 2025-07-29:
I think I might also be encountering this issue. I have a wrapper application that calls the ReadTags function from the driver to read a large amount of tags (~400,000) from a Siemens1500 PLC. The reading goes fine until for some reason the PLC decides to terminate the connection, leading to a timeout (see image).

This might be a different issue entirely, the only reason I suspect it might be realted is because I'm using the ReadTags function read chunks of 100 tags at a time (Originally I had one massive ReadTags call but I tried chunking things to see if it could solve the timeout issue) and the timeout is always encountered when the chunk contains large string tags like what's being described here

[image]

Stiwa-PohG, 2025-07-31:
I tried this on a different hardware and got the same error.

I am using TIA V20, PLCSim Advanced, without nettoplscim

I have no idea whats the problem is!

[test_read_20_String_254Char_2.zip](https://github.com/user-attachments/files/21526289/test_read_20_String_254Char_2.zip)

thomas-v2, 2025-08-01:
I did the test now with a 1512SP-1 PN with firmware 2.9, and all is working fine.

For me it looks like a bug in PLCSim Advanced.

bgattinger, 2025-08-08:
Done some more of my own testing. I'm using a siemens 1500series PLC. Specifically the CPU is a 1512SP-1 PN allowing 100 items per read request. What I have observed:

- I can read up to x3 String[254] tags before the PCL Terminates the TCP connection and timeout in the driver occurs
- x9 String[100] tags before PLC terminates connection and driver times out
- x14 String[50] tags before PLC terminates connection and driver times out
- x62 String[25] tags before PLC terminates connection and driver times out

thomas-v2, 2025-08-08:
What's the firmware version of the CPU and which TIA portal version are you programming with? As I have the same model here for testing, we can see where's the difference.

bgattinger, 2025-08-08:
> What's the firmware version of the CPU and which TIA portal version are you programming with? As I have the same model here for testing, we can see where's the difference.

CPU Firmware version is V4.0.0
and TIA portal project version is V20.0.0

bgattinger, 2025-08-08:
So it would seem that something in the V4.0.0 firmware that I had on my 1512SP-1 PN CPU was the culprit. 

I updated the firmware to V4.0.4 and now there are no issues. All the previous test cases that resulted in timeouts work now. And I've since tested MultiVariable requests with 100's of strings with different lengths, with some at max length (254), some at min length (1) and lengths in between and  the driver reads them all in fine now with no timeout issues

bgattinger, 2025-08-08:
Yeup Good ol Siemens 

[image]

thomas-v2, 2025-08-08:
Plcsim Advanced V7.0 with Update 1 should fix this Siemens bug as well, as they desrcibe the changed to firmware V4.0.4 with this version. If some of the Plcsim Advanced users can confirm this, I'll close this issue.